riscv/riscv-smmtt #2: Is `Svpams` an extension name?

a4lg, 2023-08-04:
As a person interested in this specification (as a developer once involved TEE), I'd like to prepare for it and found something not obvious from the documentation.

Considering from the CSR existence (16 CSRs allocated tentatively) and being dependent from `Smmtt`, `Svpams` looks like an extension.  However, the documentation does not refer `Svpams` as an extension.

If you have any free time, clarifying whether `Svpams` is an extension would be helpful.

rsahita, 2023-09-26:
Yes - Svpams is an extension specified as part of the broader "Supervisor Domains" specification. I will make that clear in the next update of the specification soon.

rsahita, 2023-10-03:
addressed in draft 1.0.2

closed 2023-10-03 by rsahita