riscv/riscv-smmtt #3: PMP as an option for supervisor domain isolation

NicholasWoodIMG, 2023-08-16:
Add PMP as an option for supervisor domain isolation:

- Add PMP as a mode in table 3.2
- When PMP mode is used, then MTT is not required and PMP is used instead for physical memory isolation between domains
- When PMP mode is used, the SDID CSR can still be used to identify the currently active supervisor domain

jackwdandrew, 2023-09-18:
Additionally, when only PMP and Bare modes are supported, mttp.MTTPN is redundant. Therefore an implementation that  can tie it to 0. Probably calling mttp.MTTPN out as WARL is sufficient.

rsahita, 2023-09-19:
Makes sense. Sub-fields of that CSR are already specified as WARL.

from 9-19 discussion - Krste's comment - will need to use SFENCE to flush TLBs with any PMP settings changes

kasanovic, 2023-09-19:
To clarify, if we did add SDID we may be able to provide more targeted SFENCE versions to reduce the microarchitectural translation/protection cache flushing  when PMP entries are swapped.

NicholasWoodIMG, 2023-09-19:
> *** CAUTION: This email originates from a source not known to Imagination Technologies. Think before you click a link or open an attachment ***

Thanks. That looks like it would be useful. Could we address that here in this document?

Another useful extension to PMP for supervisor domains would be to allow “switchable” PMP configurations indexed by SDID to avoid PMP reconfiguration on context switch. Similar to MMU configurations indexed by VMID.

rsahita, 2023-10-17:
Eckhard: Do we also need an informative section for the "no-MTT" Scenario where IOPMP is used for Access Control (instead of IOMMU-MTT)?

NicholasWoodIMG, 2023-10-17:
Table 2. Encoding of mttMODE field:

5|"No MTT"|Supervisor domain isolation not using MTT for physical memory isolation, for example (e)PMP.

-----------------------------

Rationale (in the intro section of the doc?):

"The Smmtt extension defines an SDID CSR, and a physical memory isolation mechanism for access control rules across domain boundaries. 

The SDID can be used by the Risc-V architecture to provide domain isolation guarantees beyond memory isolation. For example interrupt management, performance counters and debug controls. 

For memory access control, the Smmtt specification introduces MTT. MTT enables fine grained (page level) dynamic memory allocation. 

Other physical memory isolation mechanisms can also be used to enforce supervisor domain boundaries for use cases that do not require the full flexibility of MTT. For example, (e)PMP configurations controlled by M-mode during domain context switching. In this case the "No MTT" mode is used. The same security guarantees are provided to supervisor domains, and the SDID CSR is still valid. Only MTT is not used."

gagachang, 2023-10-18:
Hi
If Smmtt also supports M-mode only or M/U mode systems, where S-mode is absent, adding some description or diagrams for this use case is helpful for people.

rsahita, 2024-01-22:
Moved @gagachang's comment to separate issue.

Opened [PR](https://github.com/riscv/riscv-smmtt/pull/19) to address this issue.

rsahita, 2024-02-07:
addressed by #19

closed 2024-02-07 by rsahita