riscv/riscv-smmtt #10: Shareable data and SDID domains

jackwdandrew, 2023-09-27:
I think the intent of Smmtt is that data that passes PMP / MTT checks should be shared between supervisor domains based on:

`To support IO/memory sharing, a hart/device may perform accesses to memory exclusively accessible to its supervisor domain, and to memory shared globally with all supervisor domains or other specific supervisor domains. Memory sharing between supervisor domains is achieved by simply making the physical memory region accessible to the supervisor domains via the MTT structure. `

Later in the document it states that:

`Data and instructions cached for one supervisor domain shall not be accessible to another supervisor domain, so the SDID (along with the other execution context identifiers such as ASID, VMID, shall factor into the data and instruction caches tags to enforce isolation.`

I read this as even memory shared globally with all supervisor domains but cached under one supervisor domain cannot be accessed by another supervisor domain (even though PMP and MTT checks passed). Is this understanding correct and what's the rationale?

rsahita, 2023-10-03:
TODO: rewrite this section - add access-control enforcement as 1st principle, 
Add non-normative text for implementations (includes tagging, coherency, encryption etc.)

rsahita, 2023-10-12:
Any feedback on this implementation note:

Implementation note - Access-control isolation policies enforced via Smmtt for memory exclusive to a supervisor domain shall be enforced in caches as well. Implementations may enforce such access-control via the address-translation caches alone, with no additional changes on data/code cache structures, or via additional tagging of data/code cache lines if required to track additional domain metadata for memory encryption (or other scenarios).

ved-rivos, 2023-10-12:
Considering that a cache is nothing but just a copy of the data held in memory, such an implementation note may not be required. The second sentence about tagging for memory encryption is orthogonal to access control but since the sentence starts out by stating "may enforce...via" it seems like they are related. So suggest dropping the implementation note all together. When/if a later section provides more comprehensive treatment of memory encryption then it may include some implementation suggestions such as holding key ID with cache blocks or looking up a page-address->key_ID table on eviction from a cache, etc.

rsahita, 2023-10-13:
@jackwdandrew does that sound ok to you - to remove this implementation note entirely?

jackwdandrew, 2023-10-16:
My understanding is that the MTT structure does not define a shared/private attribute. So, if PMP+MTT checks pass then the data is implicitly shared between domains and can be read from a cache. As we require software to flush TLBs when switching domains, then there is no security issue.

Is this understanding correct? If so, then I agree that there is no need for an implementation note. I also think the below text can be removed?

`Data and instructions cached for one supervisor domain shall not be accessible to another supervisor domain, so the SDID (along with the other execution context identifiers such as ASID, VMID, shall factor into the data and instruction caches tags to enforce isolation.`

rsahita, 2023-10-25:
closing this issue (assuming @jackwdandrew ok with the redaction of the text in v1.0.3)

closed 2023-10-25 by rsahita

jackwdandrew, 2023-11-16:
All good - thanks @rsahita