riscv/riscv-smmtt #22: Moving the ISA part of external debug security to smmtt

AoteJin, 2024-01-31:
Just as discussed in external debug TG meeting on Jan 30th, it is agreed to move the ISA part to smmtt spec. File the issue to track progress.

I think we need a chapter in smmtt to fit in and rephrase this part of external debug security draft
https://github.com/joxie/riscv-external-debug-security/blob/dev/chapter2.adoc

AoteJin, 2024-02-15:
The message I received is different. I thought we agreed on moving ISA part to smmtt TG. @joxie and @gokhankaplayan to correct me.

Overall I think it is not a good idea to have two place separately document the ISA part of external debug security.

rsahita, 2024-02-20:
it is natural to have common ISA parts required for external debug be in the external debug spec. We should add any extensions needed for supervisor domains in the supervisor domains spec. hope that makes sense.

closed 2024-02-23 by rsahita