riscv/riscv-smmtt #23: TOCTOU mitigation of debug control check

AoteJin, 2024-01-31:
Previously we did not mandate the implementation details of debug control logic. The logic might be far away from the hart (e.g. inside Debug Module) depending on implementation.  This choice of design will lead to TOCTOU issue. For example, an incoming haltreq passes the check logic, but the hart changes to a undebuggable privilege mode when haltreq arrives hart. 

[image]

Mitigation: the spec shall explicitly mandate implementations to guarantee no privilege change between debug request check and serving the debug request by hart.

Similar, the same rule should be mandated for trigger that privilege must not change during debug control check and trigger firing/matching when trigger action is configured to enter Debug Mode.

rsahita, 2024-02-07:
this seems to be a micro-arch/implementation recommendation and probably applies to many such scenarios - perhaps we capture this in the RVI security model? @NicholasWoodIMG. additionally it can be capture in the non-isa spec.

NicholasWoodIMG, 2024-02-07:
Agree this does not seem specific to supervisor domains. 

It feels too "low level" for the security model. As this is specific to debug, then raising it with the TG for domain aware external debug would probably make more sense. Incidentally, I don't think this is a problem in the current external debug proposals. And either way should be easily dealt with by implementations.

ved-rivos, 2024-02-07:
Agree. This is too low level for security model. This does not seem unique to this case either. For instance, privilege being made visible to the functional unit that implements PMP functionality and if the PMP unit does not see the privilege change in time then it could lead to a bypass of the PMP. Another instance, is the halted signal going to the trace encoder not being observed by trace encoder in time to stop the trace.

rsahita, 2024-02-20:
closing this per the discussion above.

closed 2024-02-20 by rsahita