riscv/riscv-smmtt #24: Protections on external trigger

AoteJin, 2024-01-31:
External trigger missing M/S/VS/VU enable bits in tmexttrigger CSR. The debugger with S-mode privilege might affect  M-mode by enabling an external trigger and letting it fire in M-mode code to achieve attack (e.g. DoS, side channel).
Mitigation: Add per privilege enable bit in tmexttrigger like other triggers.

External trigger inputs might have resource belongs to M-mode (e.g. interrupts or performance counter etc.). A S-mode debugger might snoop the behavior the M-mode by enable those external trigger.
Mitigation: Since M-Mode is enabled by RoT, external triggers should be enabled by RoT as well .

ved-rivos, 2024-02-07:
Firstly there are no external triggers defined currently. When a external trigger facility does get defined - e.g., HPM counters overflow causing an external trigger - those facilities need to have privilege specific enables. M-mode also has control on external triggers not firing in M-mode using `MTE` control.

rsahita, 2024-02-20:
closing per above discussion - please move this to the secure ext debug TG github to track.

closed 2024-02-20 by rsahita