riscv/riscv-smmtt #33: Debug access control for U-mode only domain

AoteJin, 2024-02-23:
The current proposal has only a knob in CSR to enable debuggability for supervisor mode. It will result in privilege escalation if a domain has  only U-mode application (e.g. PENGLAI TEE architecture). This choice of U-mode only domain might reduce overhead of context switch (without save/restore S-mode context). 

[image]

There are tow direction to mitigate:
1. In addition to S mode knob, add another U mode knob.
2. Mandate that the SW has to add a thin layer of S mode runtime to enable self-hosted debug

_Originally posted by @AoteJin in https://github.com/riscv/riscv-smmtt/issues/11#issuecomment-1918645811_

ved-rivos, 2024-02-23:
A supervisor domain should not be conflated with supervisor mode. In a M+U implementation, the supervisor domain only has U-mode. In a M+S+U implementation, it is M mode policy whether it allows execution in S-mode. The external debug control is for a Supervisor Domain - and not for S-Mode in a Supervisor Domain. When a Supervisor Domain is external debug allowed, all privilege modes in that Supervisor Domain are external debug allowed.

AoteJin, 2024-02-23:
>When a Supervisor Domain is external debug allowed, all privilege modes in that Supervisor Domain are external debug allowed.

Yes, this is exactly what I understood. But this will be an issue for TEE design like this: https://www.usenix.org/system/files/osdi21-feng.pdf

This TEE architecture has M+S+U privilege mode and creates U-mode only enclave to boost performance and improve memory utilization.

rsahita, 2024-02-23:
To add, the paper referenced above asserts that "The secure monitor maintains all the enclave page tables" on page 279 (memory isolation among enclaves). From the sw arch diagram, secure monitor runs in M-mode. If the M-mode RDSM is itself not under external debug, the U-mode SD can be ext-debug allowed without any impact to the enclave page tables managed by the RDSM.

rsahita, 2024-03-05:
@AoteJin can i close this issue?

rsahita, 2024-03-14:
closing this one - please re-open if required.

closed 2024-03-14 by rsahita