riscv/riscv-smmtt #37: Is Smsdia depends on AIA for RV32 ?

gagachang, 2024-03-01:
Hi,

When I read section *6.4 Machine Interrupt registers (mip and mie)*, it says:

```
The Smsdia extension introduces the machine supervisor domain external interrupt-pending (MSDEIP
- bit 32) field in mip and the machine supervisor domain external interrupt-enable (MSDEIE - bit 32)
field in mie.
```

The `mip` and `mie` in Priv. Spec. is `MXLEN`, which is `MXLEN=32` on RV32, which only has bit 0 ~ bit 31.
AIA extends them to have high-half `miph` and `mieh`.

Does it mean that `Smsdia` depends on AIA ?

rsahita, 2024-03-05:
@gagachang that's correct - Smsdia is defined with reference to AIA

gagachang, 2024-03-05:
@rsahita Why?  Platform can not deploy multiple PLICs or CLICs?

ved-rivos, 2024-03-05:
APLIC, PLIC, and IMSIC are supported by Smsdia.

ved-rivos, 2024-03-05:
Implementing Smsdia requires implementing `mieh` and `miph` but does not require implementing AIA.

gagachang, 2024-03-06:
OK.
@ved-rivos Would you like to add some description about this ?
So that people can know Smsdia doesn't require AIA, but at least  `mieh` and `miph` should be implemented for RV32.

gagachang, 2024-03-06:
Just curious about why we don't assign interrupt number 16~23 for `MSDEI`.

[image]

rsahita, 2024-03-19:
FYI @gagachang PR #40

gagachang, 2024-03-20:
> FYI @gagachang PR #40

Thanks! @rsahita @ved-rivos 
The PR LGTM.

One last minor question.
There are still bit13 ~ bit15 for local interrupts to be assigned into `mip`.
Bit13 has been allocated for `Sscofpmf` extension.
Is there any chance to locate `LSDEIP` in bit14 or bit15 ?

ved-rivos, 2024-03-20:
> Is there any chance to locate LSDEIP in bit14 or bit15 ?

All bit allocations are tentative and subject to change before Freeze. The 14 and 15 are currently reserved for possible other uses.

rsahita, 2024-03-21:
closing this as PR #40 was acked