riscv/riscv-smmtt #45: Access control of CSR seed

gagachang, 2024-04-10:
Hello,

In RISC-V scalar crypto Zkr extension, it defines CSR `seed` for CPU to read entropy source.
And M-mode can further make `seed` available to less privilege modes, i.e., supervisor domain.

Is there any guide or recommendation about access control of `seed` between multiple supervisor domains?
Should we forbid all supervisor domain directly access `seed`?
I.e., supervisor domains must issue SBI, or trapping to M-mode to emulate accesses to `seed`.
Or can RDSM context switch `mseccfg.{s|u}seed` to let some of supervisor domains have access right?
But in this way, malicious SD may continuously read `seed` to launch DOS attack, am I correct?

ved-rivos, 2024-04-10:
A RDSM can delegate access to `seed` CSR to a supervisor domain. The `seed` CSR is a per hart CSR. For the duration of the SD execution, the SD can consume entropy from the `seed` CSR. There is no specific reason to disallow the delegation. Consuming entropy using `seed` CSR from one hart should not affect the entropy availability on another hart. Please elaborate more on the DOS attack concern.

gagachang, 2024-04-11:
>  Please elaborate more on the DOS attack concern if it is beyond the shared entropy source concern.

Let's say
1. SD(a) is going to execute csrrw instruction to read `seed`.
2. Suddenly SD(b)'s interrupt asserted, causing CPU traps to RDSM.
3. RDSM immediately performs context switch for this hart, saving SD(a)'s context and restoring SD(b).
4. However, SD(b) reads `seed` in its interrupt handler. SD(b) may consume much of entropy.
5. SD(b) finishes its handler. RDSM performs context switch back to SD(a). SD(a) executes previous csrrw, but it might not get sufficient entropy.

Above case 1 to case 5 can be an infinite loop.

ved-rivos, 2024-04-11:
Got it. When such a bad entropy source is implemented, the RDSM as a policy should not delegate `seed` to SDs.

ved-rivos, 2024-04-11:
Depends on quality of entropy source and the SW policies. The ISA provides controls to support either policies than an RDSM may want to implement.

gagachang, 2024-04-11:
> Depends on quality of entropy source and the SW policies. The ISA provides controls to support either policies than an RDSM may want to implement.

It makes sense.
Seems we don't need any ISA changes for access control of `seed`.
Thanks!

closed 2024-04-11 by gagachang