riscv/riscv-smmtt #51: Why force SDID to zero in Bare mode?

eckhard-delfs-qualcomm, 2024-06-05:
Chapter 3 outlines constraints to the `MTTP` register when setting `MODE`=`Bare`: 
_In this case, the remaining fields (`SDID`, `MTTPPN`) in mttp must be set to zeros, else generate a fault_.

Is there an architectural reason why the `SDID `field needs to be forced to a fixed value in this case? Can it not be set arbitrarily?
If an architecture wants to use only Smsdid (without Smmtt) and enforce domain isolation via PMP/IOPMP, then there would be no tangible qualifier in the architecture to differentiate different SDs.

One scenario may be a M+U architecture where different U-mode-only SDs may need to be protected by dedicated trace/debug permissions. In todays' spec version that is supported via msdcfg settings, but in this case there would not be any ISA-level visible identifier to distinguish.

Another scenario is a deployment with Smmtt, where a given SD shall be transitioned to/or from `Bare` mode. If there is a transition to `Bare` mode, then this implies an `SDID `switch from xyz to zero. In case of multiple domains applying that use case, they would all collapse to `SDID `= 0, even if RDSM enforces PMP isolation between them.

In short, if there is no stringent reason to enforce `SDID=0` for `Bare` mode applications, I would suggest removing that requirement.

gagachang, 2024-06-11:
@eckhard-delfs-qualcomm
Chapter 2 says:
The SDID is a local identifier for the hart and may be used to tag hart-local resources to access-control data associated with the supervisor domain.

For the peripherals on the bus fabric, e.g., IOPMP, they need "global" ID to distinguish different SDs.
Worldguard should take care on global IDs tagged into bus transactions.

rsahita, 2024-06-20:
@eckhard-delfs-qualcomm So the SDID tag only makes sense when the MTTP is valid and mode is !BARE. since the only use of it is to facilitate address-protection fences on a per supervisor-domain basis; The analogy is the VMID and HGATP (priv ISA manual section 18.2.10). SDID does not modify any PMP behavior.

eckhard-delfs-qualcomm, 2024-06-20:
@rsahita: Thanks for providing the analogy with G-stage hgtap! Makes sense. 
Closing the issue.

closed 2024-06-20 by eckhard-delfs-qualcomm