riscv/riscv-smmtt #59: Question about siselect when SDICN is zero

gagachang, 2024-07-28:
Current Smsdia (section 6.1) states:

```
When SDICN is not the number of an implemented interrupt controller, or is zero then the following
rules apply to all privilege modes:

• All siselect values that access IMSIC registers are reserved.
```

What does "reserved" mean ?
Shouldn't it raise an illegal instruction exception in this case ?

Recommended changes if it should raise an illegal instruction exception:
```
When SDICN is not the number of an implemented interrupt controller, or is zero then the following
rules apply to all privilege modes:

• When siselect is a number in the range 0x70–0xFF (to access IMSIC registers), attempts to access
sireg raises an illegal instruction exception.
```

Therefore, RDSM can emulate an IMSIC when SDICN is set to 0.

gagachang, 2024-07-28:
Hello @ved-rivos , would you like to give some perspectives ?

ram2532, 2024-09-14:
I don't think the spec intends us to infer that the behavior defined for SCICN=0 or an unimplemented interrupt controller is meant to enable emulation. Rather, it's causing the hart to behave like Supervisor and VS guest external interrupts do not exist.

Regarding what reserved means, the RISC-V Instruction Set Manual Volume I Unprivileged Architecture States the following under section 2.2. Base Instruction Formats:

> The behavior upon decoding a reserved instruction is UNSPECIFIED.

I would think a reserved CSR encoding within a CSR instruction would make that instruction UNSPECIFIED.

The AIA specifications recommends taking an illegal instruction exception -

> When siselect is a number in a reserved range (currently 0x00–0x2F, 0x40–0x6F, or a number above 0xFF not designated for custom use), or in the range 0x70–0xFF when there is no IMSIC, attempts to access sireg should preferably raise an illegal instruction exception (unless executing in a virtual machine, covered in the next section).

So, by defining the IMSEC register range as reserved, the spec leaves it up to the implementation to handle it in the way the implementation handles all other reserved instructions.

But, we do need clarification from @ved-rivos or @rsahita

gagachang, 2024-09-19:
> So, by defining the IMSEC register range as reserved, the spec leaves it up to the implementation to handle it in the way the implementation handles all other reserved instructions.

If the behavior is implementation-defined, I have no more doubts.
Thanks.

closed 2024-10-01 by rsahita

gagachang, 2024-10-02:
@ved-rivos  It makes sense. Thanks!

PR #79 (2024-10-01): clarify behavior of invalid SDICN
closes #59