riscv/riscv-smmtt #60: Question about SSM effect in CBQRI section

yong-xuan, 2024-08-09:
Hi @ved-rivos,

It seems that the description of SSM is reversed with r/w value computing process and given example.

srmcfg.RCID is allowed SD to configure lower order SRL bits when SSM is 0.
> When SSM is 0, the SRL configuration enables the RDSM to configure a RCID prefix value in the srmcfg.RCID field while allowing the supervisor domain to program the low order SRL bits of the field.

But in r/w value computing process, SD can access lower order SRL bits when SSM is 1.
> SRL_MASK = (1 << SRL) - 1
> if SSM == 1
>     RCID-value = srmcfg.RCID & SRL_MASK

> if SSM == 1
>    srmcfg.RCID = (srmcfg.RCID & ~SRL_MASK) | (RCID-value & SRL_MASK)

and also the example
> The RDSM configures SRL to 3 and SSM to 1 for the second SD and programs the srmcfg with a value of 24. The second SD is allowed to program the low order 3 bits of srmcfg.RCID

ved-rivos, 2024-10-01:
Thanks. Yes, the listing should have used "SSM == 0" instead of "SSM == 1". PR #67 fixes it.

rsahita, 2024-10-01:
PR #67 was merged - closing.

closed 2024-10-01 by rsahita