riscv/riscv-smmtt #61: Details of Exception Reporting

sqzsq, 2024-09-09:
The spec currently specifies that an access fault is raised whenever the type of access is denied by the current configuration in the MTT. Providing more details about the access fault may be useful, such as the address of the memory access.

In particular, indicating a violation resulted from MTT check may help distinguish it from other access faults resulted from PMP or PMA. Does MTT requires the H extension? If it does, maybe mtval2 can be used for this purpose, since mtval should be used for the address of the memory access.

rsahita, 2024-09-09:
so MTT is evaluated in parallel with PMP, so the mcause will be the apropos access-fault for any MTT permission violations. That is identical to PMP or PMA violations. That seems sufficient for the M-mode sw to resolve an MTT access-fault - also note that such violations are expected to be fatal and non-recoverable.

sqzsq, 2024-09-11:
It's true MTT access faults are non-recoverable. Still if I recall correctly, PMP and MTT plays different roles. PMP is for restricting access to the entire physical address space including MMIO, while MTT is meant for just the DRAM space. They are setup by different components and the exceptions may be of interest to different components. Say a device driver caused a PMP violation because it crossed the MMIO space, that is different from a MTT violation caused by some buggy TVMs. If it's the former case, there's no TVM to kill.

rsahita, 2024-10-01:
MTT can cover MMIO space as well (in fact with CoVE-IO like usages, it will be used to cover MMIO for the IOMMU and devices assigned to a Supervisor domain.  From M-mode perspective, a MTT or PMP violation is non-recoverable for the workload - a driver causing an MTT or PMP fault cannot be differentiated as buggy or malicious., it's no different than software trying to access invalid memory regions.

closed 2024-10-01 by rsahita