riscv/riscv-smmtt #62: Why not have fixed sized MMT tables at all levels ?

avpatel, 2024-09-09:
The MTT tables at different levels are not of same size and MMT L2 is particularly very large. This makes memory management for MTT tables more involved. Instead of this, if all MTT tables were of same size (just like MMU or IOMMU page tables) then it will be very easy to manage a pool of fixed size MMT tables.

rsahita, 2024-09-09:
this was discussed very early on in the TG - the MTT structures are biased towards minimizing lookups, and being compact. Also important is to support all arch page sizes. Another key SW aspect is top level L3, L2 MTT structures will need to be initialized early on in system initialization by the RDSM - lower level structures L1 are 4KiB based so as differing permissions are required, large pages can be fragmented easily and can L1 table pages can be flexibly allocated from pool of 4KiB pages.

rsahita, 2024-10-03:
@avpatel I added a check for pa width in the MTT walk - ptal at PR #88 
This should address your concern with overcommitting space for L3 and L2 tables.
Separately I will also send a PR for adding a type field in L3 which will allow for further flexibility.

avpatel, 2024-10-04:
Thanks, this looks reasonable for platforms with smaller system address space.

rsahita, 2024-10-07:
PR #88 addresses this issue so closing this one.

closed 2024-10-07 by rsahita