riscv/riscv-smmtt #63: Contradiction in `mttp` description

SiFiveHolland, 2024-09-11:
In section 3, the description of `mttp` includes the sentence "In this case, the remaining fields (`SDID`, `PPN`) in `mttp` must be set to zeros, else generate a fault." This statement contradicts the register diagram just above that paragraph, which describes these fields as WARL.

eckhard-delfs-qualcomm, 2024-09-11:
Maybe rephrase according to priv spec 10.1.11 (no fault generation):
_To select MODE=Bare, software must write zero to the remaining fields (`SDID`, `PPN`) in `mttp`.
Attempting to select MODE=Bare with a nonzero pattern in the remaining fields has an UNSPECIFIED effect on the value that the remaining fields assume and an UNSPECIFIED effect on supervisor domain protection behavior._

ram2532, 2024-09-12:
I agree with @eckhard-delfs-qualcomm 

The mttp is similar in definition and behavior as to satp so I'd prefer similar write behavior when MODE=Bare. In particular, I don't think it should take an exception.

rock-ifly, 2024-10-11:
I don't think it should take an exception when mttp.mode=Bare while ((mttp.sdid != 0) | (mttp.ppn !=0)). Since we can just bypass the mttCheck when mttp.mode-Bare, just neglect mttp.sdid and mttp.ppn.

On one hand, the WARL should not raise an exception as spec says. On the other hand, we can do the same thing as satp. The satp won't throw an exception when (satp.mode =Bare) & ((satp.asid != 0) | (satp.ppn != 0) ).

closed 2024-10-14 by rsahita

rock-ifly, 2024-12-10:
Q1: Are there any comments on the following description on mmpt? What does the "not ignored" mean, report a fault? if so, what kind of the fault, illegal instruction fault?
> Implementations are not required to support all defined MODE settings when XLEN=64. A write to mmpt
with an unsupported MODE value is not ignored. Instead, the fields of mmpt are WARL in the normal
way, when so indicated in Register 1 and Register 2.

Q2: The WARL field should keep the original value if unsupported value is written? does this mean that if a write to mmpt with an unsupported mode should be ignored?
> RISCV privilledge vol2: Implementations will not raise an exception on writes of unsupported values to a WARL field.