riscv/riscv-smmtt #65: Review comments/questions on Smsdia v0.1

SiFiveHolland, 2024-09-11:
* Neither section 3.1 nor section 6.1 defines a reset value for `msdcfg.SDICN`. Having the reset value be zero would break non-Smsdia-aware software, so I would suggest specifying a non-zero reset value (unless the field is read-only zero).
* Section 6.1 describes SDICN as a WARL field, which makes the condition "When `SDICN` is not the number of an implemented interrupt controller, or is zero" redundant. Any value larger than the maximum implemented supervisor domain interrupt controller number would not be legal, right?
* The privileged specification defines `mip.SEIP` as writable by software. It should remain writable by software when `msdcfg.SDICN` is 0. The wording "... `mip.SEIP` is 0" seems to contradict with this.
* Section 6.4 prescribes a priority order for major interrupts. However, the AIA provides a way for software to change these priorities. I would recommend to clarify that the order in section 6.4 is the _default_ order.
* Has there been any consideration to excluding the bit of `msdeip` corresponding to `msdcfg.SDICN` when calculating the value of `mip.LSDEIP`? In other words, always exclude the current supervisor domain from triggering LSDEI. This could avoid the need to context switch the `msdeie` register.
* For supervisor timer interrupts (including with Sstc), is the assumption that the RDSM will program an M-mode timer based on the aggregate `stimecmp` values from all inactive supervisor domains? And since LSDEI cannot be triggered by software, the RDSM would require some other method (SBI SSE, a virtual interrupt) to notify S-mode that another supervisor domain has a pending timer interrupt?

ved-rivos, 2024-10-01:
1. Machine mode is required to initialize machine state, including `msdgcfg.SDICN`. The architecture specifies a reset value only if  such state affects machine mode execution. For state that does not affect machine mode execution out of reset, the reset state is unspecified.
2. Section 6.1 is incorrectly stating SDICN as WARL. It should be WLRL. Fixed in #80 
3. The SEIP is a logical OR of the software writeable SEIP bit and the signal from an interrupt controller indicating the external interrupt is pending. The wording specifically refers to the signal and not the software writable SEIP bit - "S-level external interrupt pending signal indicated in `mip.SEIP` is 0" and not that SEIP is read-only zero.
4. Agreed. Updated in https://github.com/riscv/riscv-smmtt/pull/81
5. A context  switch of msdeie is not required as it is not volatile. A write to msdeie is required. The choice was guided by two considerations a) avoid complexity in hardware to do the masking while considering SDIC being WLRL b) not preclude use cases where the RDSM may want to be notified of interrupts pending for the current active supervisor domain - such as for instance to address threats such as SGX-step.
6. It is not required for  RDSM to notify the current active supervisor domain that a timer interrupt is pending for another supervisor domain. That may be considered as a side/covert channel in many designs. However, such use is not precluded. When the RDSM does intend to notify a supervisor domain about timer  interrupts or any other interrupts pending  for another  supervisor domain then  it  may  use methods  such as a virtual  interrupt or an  SSE.

SiFiveHolland, 2024-10-02:
Thanks for the clarifications!

> 1. Machine mode is required to initialize machine state, including msdgcfg.SDICN. The architecture specifies a reset value only if such state affects machine mode execution. For state that does not affect machine mode execution out of reset, the reset state is unspecified.

This ambiguity requires M-mode software to add explicit support for this ISA extension to allow running on hardware that implements the extension, even if that M-mode software has no desire to use the functionality added by the extension. I'm not aware of any other ISA extensions with this requirement.

> 2. Section 6.1 is incorrectly stating SDICN as WARL. It should be WLRL. Fixed in https://github.com/riscv/riscv-smmtt/pull/80

Is the expectation that the platform will rely on DT or equivalent to discover the valid SDICN values (and the mapping between SDICN values and MMIO ranges)?

> 5. A context switch of msdeie is not required as it is not volatile. A write to msdeie is required. The choice was guided by two considerations a) avoid complexity in hardware to do the masking [...]

Makes sense -- though for b) isn't that just as easily achieved by clearing the SEI bit in `mideleg`? A single interrupt triggering both SEI and LSDEI seems redundant.

ved-rivos, 2024-10-10:
I have updated PR #93 to provide a reset but the reset default is platform specific. 

> Is the expectation that the platform will rely on DT or equivalent to discover the valid SDICN values (and the mapping between SDICN values and MMIO ranges)?

DT and ACPI are more appropriate for lower privileges. Machine mode is usually what creates the DT or ACPI tables for consumption by lower levels. The machine mode is expected to know the machine. It may rely on low level discovery methods such as unified discovery.

rsahita, 2024-10-10:
last part addressed by #93