riscv/riscv-smmtt #68: v0.1 Review Comments/Questions: Regarding physical address ranges identified by mttp MODE values

ram2532, 2024-09-12:
Section 3 "Smsdid: Supervisor Domain Identifier and Protection Register" defines valid mttp MODE values. In particular, for RV64, it defines Smmt46 and Smmt56 and for RV32, Smmt34.

 When I read this section, I take these values to identify the maximum possible physical address size, but that the implemented physical address (PA) could be smaller. To be specific, Smmt46 identifies that the implemented PA size does not exceed 46 bits. If the actual PA size is less that that, then those upper bits will always be 0. Further, Smmt56 identifies that the implemented PA size is greater that 46 bits but does not exceed 56 bits.

However, I wonder if the tables should explicitly state this fact.

For example,
For Smmt46: "Page-based supervisor domain protection for physical addresses less than or equal to 46 bits with RWX permissions per page"
For Smmt56: "Page-based supervisor domain protection for physical addresses between 47 and 56 bits with RWX permissions per page"

Further, what if the PA size is > 56 (for RV64)? While the RISC-V virtual memory specification restricts PA available for translation to 56 bits, if the translation MODE is set to Bare or there is no virtual memory, a supervisor/user could address a PA of up to 64 bits. I'm not sure who would do that or why.

rsahita, 2024-10-09:
@ram2532 I think this is addressed by PR #88 - PTAL and confirm

ram2532, 2024-10-10:
@rsahita
To confirm my understanding,

For a given MTT mode (Bare, Smmtt34, Smmtt46, Smmtt56), the maximum addressable physical address (**MAPA**) is the smaller of the physical memory  size specified by 

- The platform when MMT mode is Bare or
- The current MTT mode (34b for Smmtt34, 46b for Smmtt46 or 56b for Smmtt56).

Is that correct?

I think some language would be useful that explicitly says that an MTT mode specifies the maximum addressable physical address - i.e. Smmtt34 mode specifies 34b MAPA, Smmtt46 specifies 46b MAPA, Smmtt56 specifies 56b MAPA.

SiFiveHolland, 2024-10-10:
> Smmt56 identifies that the implemented PA size is greater that 46 bits but does not exceed 56 bits.

This is not correct. Smmtt56 can be used on platforms with any physical address width. If the PAW is 46 bits or smaller, the hardware will only access the first MTTL3 entry.

> * The platform when MMT mode is Bare or

The platform-defined limit applies for all active MTT modes (Smmtt34/Smmtt46/Smmtt56), in addition to the limit defined by the MTT layout.

ram2532, 2024-10-10:
@SiFiveHolland My latest remarks are with respect to the changes made in PR #88. I'm no longer asking for the changes in my original post.

rsahita, 2024-10-14:
@ram2532 Please see if this clarification addresses your question (PR #97 )

closed 2024-10-15 by rsahita