riscv/riscv-smmtt #72: QC v0.1 review comments

eckhard-delfs-qualcomm, 2024-09-16:
- Chapter 3.2 M-mode Supervisor Domain Fence Instruction
This section needs more clarity on the precise operation of the fencing instruction when any of {rs1, rs2} is/are not zero. 

- Chapter 5.6 Operand 1 register
The section ends with statement: "_If the initial low-order 0 bit position is denoted as x, the size of the range is computed as (1 << (12 + x + 1))._ We think it would help to add an example which clarifies the relation between range size and PPN field encoding.

- Chapter 6: Smsdia: Supervisor Domain Interrupt Assignment
Statement: _The RDSM can employ the MTT and/or PMP to limit a supervisor domain’s access to the memorymapped
register interface of the interrupt controller associated with it._ 
The chosen term "_limit_" is ambiguous, as it does not indicate the type of limitation. Maybe: _The RDSM can utilize the MTT and/or PMP to map the memory-mapped register interface of an interrupt controller exclusively to a single supervisor domain, preventing access by other domains._

- Chapter 8.4 Supervisor Domain QoS Register Interfaces (QRI)
The fifth paragraph states: "_The CONFIG_QRI_LIMIT operation may be requested once following reset_."
What is the reset value of NCBLKS for a QRI if the CONFIG_QRI_LIMIIT operation is not performed for that QRI? Is this implementation defined?

ved-rivos, 2024-10-01:
> Last paragraph. It is not specified if switching mttp.MODE from Bare to any other and vice versa mode takes effect immediately or requires execution of an MFENCE.SPA instruction.

A MFENCE.SPA should be required.

ved-rivos, 2024-10-01:
> This section needs more clarity on the precise operation of the fencing instruction when any of {rs1, rs2} is/are not zero.

I think the following text could be added:

- If rs1=x0 and rs2=x0, the fence orders all reads and writes to the MTT for all supervisor domain address spaces.
- If rs1=x0 and rs2!=x0, the fence orders all reads and writes to the MTT for the supervisor domain address space identified by the SDID in rs2.
- If rs1!=x0 and rs2=x0, the fence orders all reads and writes made to the MTT that correspond to the physical address in rs1, for all supervisor domain address spaces.
- If rs1!=x0 and rs2!=x0, the fence orders all reads and writes made to the MTT that correspond to the physical address in rs1, for the supervisor domain address space identified by the SDID in rs2.

eckhard-delfs-qualcomm, 2024-10-02:
> I think the following text could be added:
>
> * If rs1=x0 and rs2=x0, the fence orders all reads and writes to the MTT for all supervisor domain address spaces.
> * If rs1=x0 and rs2!=x0, the fence [...]

Looks good to me. Thanks!

eckhard-delfs-qualcomm, 2024-10-02:
> > Last paragraph. It is not specified if switching mttp.MODE from Bare to any other and vice versa mode takes effect immediately or requires execution of an MFENCE.SPA instruction.
> 
> A MFENCE.SPA should be required. The last paragraph specifies that "even if the old or new mode is Bare"

Correct. Did not spot that statement before.

andrewdellow, 2024-10-02:
> > The chosen term "limit" is ambiguous, as it does not indicate the type of limitation.
> 
> The term "limiting access" is well defined. But to avoid confusion it is reworded as "restrict a supervisor domain's access" in PR #83.
> 
> While use of PMP to restrict the access to exclusively to a supervisor domain is the expected use case, the PMP or MTT by itself does not enforce such an exclusivity - a RDSM may allow access from multiple supervisor domains using suitable PMP/MTT configurations.

The PMP and MTT by themselves don't enforce anything other than the intention of the RDSM, so is the intention here to provide a suggested use case, a recommendation, a definition ?  maybe it is better to say _the RDSM can employ the MTT and/or PMP to restrict a supervisor domain’s access to the memory mapped register interface of an interrupt controller, for example the RDSM may limit access to the memory-mapped register interface of an interrupt controller exclusively to a single supervisor domain, preventing access by other domains_

ved-rivos, 2024-10-02:
I agree that this sentence is not material to the specification. It's stating the obvious for what an RDSM might do. I will remove the sentence.

eckhard-delfs-qualcomm, 2024-10-02:
All comments addressed - thanks, @ved-rivos

closed 2024-10-02 by eckhard-delfs-qualcomm