riscv/riscv-smmtt #76: Questions about SDIDs

SiFiveHolland, 2024-09-17:
Due to the fact that SDIDLEN is small (allowed to be as low as 0), and the only standard use of SDIDs is for fences, it appears that SDIDs are intended to be a short-lived identifier, not remaining stable over the life of a supervisor domain.

* Is this correct? If so, I would suggest the specification explicitly state that SDIDs are not a stable identifier.
* Is it expected that the same supervisor domain may have different SDIDs across harts or I/O MTTs?
* Is it allowable for the platform to tag bus transactions (e.g. NS bit or equivalent) with the current SDID or some subset of it?

rsahita, 2024-09-21:
that is not the case - SDID is a hart-local identifier, in the sense, the RDSM can activate a SDID (and MTT) on a specific hart which another hart has a different SDID (and MTT) active. For most typical usages, the SDID will remain stable over the life of the supervisor domain.  It is allowed for a platform to tag bus transactions with current SDID (and other IDs) - for e.g. to drive downstream crypto context selection (but that is implementation-specific).

I can add this into the non-normative/informative section of the spec.

ved-rivos, 2024-10-02:
SDID should not attempt to solve the ID on the bus problem. There is a fast track extension underway to define a world ID for that purpose: https://drive.google.com/drive/u/0/folders/1PGDrAZoAc6qin1jqxl6TCDFWIJtP5I56

rsahita, 2024-10-02:
@SiFiveHolland Any other changes to PR #85 ?

closed 2024-10-02 by rsahita

SiFiveHolland, 2024-10-02:
I am assuming from your response that statement 2 is correct (and statement 1 is not). So we shouldn't suggest mapping the SDID to anything outside the hart.