riscv/riscv-smmtt #77: Effects of Smsdid on existing fence instructions

SiFiveHolland, 2024-09-17:
There is a hierarchy of ASID < VMID < SDID. The H extension specification limits the scope of the SFENCE.VMA instruction to a single VMID:

> When V=1, the virtual-address argument to SFENCE.VMA is a guest virtual address within the current virtual machine, and the ASID argument is a VS-level ASID within the current virtual machine. The current virtual machine is identified by the VMID field of CSR hgatp, and the effective ASID can be considered to be the combination of this VMID with the VS-level ASID. The SFENCE.VMA instruction orders stores only to the VS-level address-translation structures with subsequent VS-stage address translations for the same virtual machine, i.e., only when hgatp.VMID is the same as when the SFENCE.VMA executed.

Will the SDID similarly be used to limit the scope of other SFENCE.* and HFENCE.* instructions?

rsahita, 2024-09-21:
good point - how is this behavior modification spec:

The behavior of the SFENCE.VMA and the HFENCE.VMA instructions are affected when
supervisor domains are used by the M-mode RDSM (per the current SDID).

When SFENCE.VMA is used within a supervisor domain, the virtual-address argument
is a virtual address with either the ASID being a S/HS-level ASID (V=0), or a
VS-level ASID (V=1).

For S/HS-level ASID, the virtual-address argument to SFENCE.VMA is a host
virtual address within the current supervisor domain, and the ASID argument is
a S/HS-level ASID within the current supervisor domain. The current supervisor
domain is identified by the SDID field of the CSR mttp, and the effective ASID
can be considered the combination of the SDID and the S/HS-level ASID. The
SFENCE.VMA orders stores only to this S/HS-level address-translation structures
with subsequent HS-level address translations.

When V=1, the virtual-address argument to SFENCE.VMA is a guest virtual address
within the current virtual machine, and the ASID argument is a VS-level ASID
within the current virtual machine. The current virtual machine is identified by
the SDID field of the CSR mttp, the VMID field of CSR hgatp, and the effective
ASID can be considered to be the combination of this SDID and VMID with the
VS-level ASID. The SFENCE.VMA instruction orders stores only to the VS-level
address-translation structures with subsequent VS-stage address translations for
the same virtual machine, i.e., only when mttp.SDID and the hgatp.VMID is the
same as when the SFENCE.VMA executed.

rsahita, 2024-10-01:
@SiFiveHolland ptal PR #84  thanks

SiFiveHolland, 2024-10-02:
Thanks, further comments are in #84.

rsahita, 2024-10-09:
PR #84 is reviewed and merged.

closed 2024-10-09 by rsahita