riscv/riscv-smmtt #78: Can the RDSM be put in HS mode?

SiFiveHolland, 2024-09-17:
For systems which implement the S and H extensions, Smmtt and I/O MTT add a third level of page table walks for memory accesses. This will have a runtime performance impact. An alternative to adding more page table walks would be to use the H extension to implement the RDSM functionality: the RDSM would run in HS-mode, and emulate the H extension to allow existing supervisor software (including hypervisors) to run unmodified in VS-mode. This would use G-stage page tables to implement domain isolation, and VS-mode IMSIC register files to provide efficient interrupt handling across supervisor domains. This would require fewer ISA changes and would possibly be more performant.

If using nested virtualization to implement supervisor domains is estimated to be less performant than Smmtt, new ISA functionality could be added to accelerate nested virtualization. Such new ISA functionality would benefit all uses of nested virtualization, not just supervisor domains.

rsahita, 2024-09-21:
As we discussed at Linux Plumbers recently, this has been documented as a TSM as the root hypervisor deployment model in the CoVE spec - please see: https://github.com/riscv-non-isa/riscv-ap-tee/releases/download/v0.7/riscv-cove.pdf
As specified in that deployment model, Smmtt is not required. In other deployment models, where the data center operator hypervisor retains HS (root) operation, the TSM is run as a peer HS root to enforce security properties for the TVMs under its purview - this model may use Smmtt if it requires scalable physical memory isolation between the host and secondary supervisor domains.

rsahita, 2024-10-01:
closing - please renew if any additional clarification needed.

closed 2024-10-01 by rsahita

SiFiveHolland, 2024-10-02:
No, I am satisfied as long as software interfaces like CoVE do not add a hard dependency on Smmtt for any of their functionality.