riscv/riscv-smmtt #100: Encoding of INFO for 2M_PAGES and 4M_PAGES

pright, 2024-10-18:
The spec says, 

For 2M_PAGES,
> The INFO field holds 16 2-bit PERM access-encoding for each 2 MiB address range to indicate ...
> Bits mpte.info[31:0] holds the 2 MiB page permissions, and mpte.info[43:32] are reserved and must be zero.

and for 4M_PAGES,
> The INFO field holds 8 2-bit PERM access-encoding for each 4 MiB address range to indicate ...
> Bits mpte.info[15:0] holds the 2 MiB page permissions, and mpte.info[21:16] are reserved and must be zero.

However, it does not specify whether the encoding is in big-endian or little-endian order. Regarding 2MB_PAGES, whether the 2-bit PERM for the first 2MB address range is encoded in INFO[1:0] or INFO[31:30] remains unclear.

ved-rivos, 2024-10-21:
> However, it does not specify whether the encoding is in big-endian or little-endian order. 

MTT accesses are implicit accesses and all implicit accesses are little-endian - even when the hart is big-endian.

pright, 2024-10-21:
> > However, it does not specify whether the encoding is in big-endian or little-endian order.
> 
> MTT accesses are implicit accesses and all implicit accesses are little-endian - even when the hart is big-endian.

To my knowledge, the Priv spec states that implicit accesses from S-mode are controlled by SBE. If MTT accesses are classified as implicit accesses, should they also be subject to SBE control?

> For implicit accesses to supervisor-level memory management data structures, such as page tables,
> endianness is always controlled by SBE.

ved-rivos, 2024-10-25:
> If MTT accesses are classified as implicit accesses, should they also be subject to SBE control?

Yes, you are correct. These implicit accesses to the MTT that is setup by M-mode and they should be governed by MBE control.

rsahita, 2024-10-28:
@pright PTAL PR #104 addresses this issue.

closed 2024-10-28 by rsahita