riscv/riscv-smmtt #113: Question about the priority order for handling interrupts from multiple supervisor domains

xlxxxxxl, 2024-11-21:
The default interrupt handling priority described in Section 6.4. And with the Note below, RDSM will clear the bit corresponding to that interrupt controller selected by msdcfg.SDICN in msdeie prior to resuming execution of the supervisor domain.

Here I have a question:
When both a "non-active supervisor domain interrupt (LSDEIP)" and an "active supervisor domain interrupt (VSEI)" occur simultaneously, why is the default to prioritize handling interrupts from other domains rather than prioritizing TVM interrupts within the current active domain?
If currently in the confidential supervisor domain, TSM needs to perform a TEERET to the VMM, leaving the scheduling decision to the VMM.

When interrupts occur in multiple supervisor domains simultaneously, if each domain prioritizes handling interrupts from other domains, could this lead to a "domain scheduling storm" where no interrupt is actually processed to completion?

Looking forward to your answer, thank you!

rsahita, 2024-11-25:
hi @xlxxxxxl - LSDEI is higher priority following the rule that interrupts for higher privilege level have higher priority - that said, there was a bug in that part of the spec - see PR #115   - the priority order should have been: high-priority RAS event, MEI, MSI, MTI,  LSDEI, SEI, SSI, STI, SGEI, VSEI, VSSI, VSTI, LCOFI, low-priority RAS event.

Regarding your other question - no such storm/starvation is possible since the RDSM has higher priority and can choose any scheduling policy. The confidential domain scenario you mentioned is one policy model where the VMM controls the scheduling, but other models are also possible.

xlxxxxxl, 2024-11-26:
Hi @rsahita, thank you for your answer. Also please allow me to ask one more question.

The newly introduced mip.LSDEI and sip.LSDEI in smmtt spec do allow a flexible choice of handling interrupts from inactive SDs in either M-mode or S-mode. This capability comes from instantiating multiple physical IMSICs for each SD in each hart, thereby providing independent S and VS interrupt files resources for each SD.
But why are these function points not reflected in the CoVE specification? The definition of COVI's ABI in CoVE does not seem to mention the use of mip.LSDEI, sip.LSDEI, and the independent VS interrupt files that each SD has because of multiple IMSICs, as well as the related process processing.

The description I've seen in CoVE is that it looks like all SDs are using one set of VS interrupt files. VMM manages and distributes them between SDs. The guest interrupt files are converted between the non-confidential SD and the confidential SD (using MPT page table isolation) through the `tvm_vhart_imsic_bind` and `tvm_vhart_imsic_unbind` interfaces, and also migrates the interrupt status between the MRIF and the guest interrupt file. It does not reflect the use of the VS interrupt file resources in different IMSIC that are independently provided for each domain.

I'm confused as to how the functionality of mip.LSDEI and sip.LSDEI, as well as the VS interrupt file resources in multiple physical IMSICs, will ultimately be applied to resolve interrupt handling for active and inactive SDs.

Please help solve my confusion, thank you!

jimingchen111, 2024-11-27:
When considering the support for setting different numbers of VS interrupt files for different Supervisor Domians, I was confused by the definition and implementation of some parameters (e.g. GEILEN, VGEIN, hgeie, and hgeip)
The GEILEN parameter is described in the specification as follows

**For any one physical hart, the maximum number of virtual harts that may directly receive guest external interrupts is limited by GEILEN.**

This parameter appears to describe the total number of VS interrupt files for a physical hart(regardless of how many S-domains are running on that physical hart);

In the SMMTT specification, VGEIN is described as follows

**The hstatus. VGEIN selects a guest interrupt file in the selected IMSIC and vsiselect and vstopei CSRs operate on the registers of the corresponding guest interrupt file in the selected IMSIC.** 

VGEIN selects a VS interrupt file in an IMSIC on a physical hart. But for the first VS interrupt file of a certain domain, it may actually be the nth VS interrupt file on this physical hart. Should the VGEIN be assigned to the actual VS interrupt file number instead of the VS interrupt file number from a certain domain perspective?
My question is, should the parameters related to VS interrupt files (e.g. GEILEN, VGEIN, HGEIE, and hgeip) be assigned from the perspective of the domain or from the perspective of the physical hart under the smmtt specification?

Many Thanks！

rsahita, 2024-11-28:
@jimingchen111 hi, so VGEIN etc. is assigned from the perspective of the selected IMSIC (which is assigned to an SD and is in turn selected based on the interrupt controller identified by the SDICN).

rsahita, 2024-12-04:
PR #115 merged.

closed 2024-12-04 by rsahita