riscv/riscv-smmtt #122: Conflicting statements about mip.LSDEI in AIA and Smsdia spec

Baixyzz, 2025-02-05:
From AIA spec,we can see

> For whichever standard local interrupts are implemented, the corresponding bits in CSRs mip and mie must be writable.

From Smsdia spec,we can see

> The Smsdia extension introduces the local supervisor domain external interrupt-pending (LSDEI).This interrupt is treated as a standard local interrupt that is assigned to bit 16 in the mip, mie, sip, and sie registers.

> The mip.LSDEIP bit is read-only.

LSDEI is treated as a standard local interrupt, so mip.LSEIP should be writable if we follow AIA spec. But Smsdia spec proposed the opposite rule. Which one is correct?

Baixyzz, 2025-02-07:
@rsahita About this question，can you give us some advice?

rsahita, 2025-02-07:
hi @Baixyzz thanks for pointing it out - the mip.LSDEIP should be writable. I will correct in a PR.

Baixyzz, 2025-02-07:
@rsahita Thanks for your reply. The  the sip.LSDEIP has the same problem.

rsahita, 2025-03-20:
Addressed in #129