riscv/riscv-smmtt #139: question about supported smmpt34 page size

Linthereeder, 2025-04-24:
On page 19, It is written that "Any level of MPTE may be a leaf MPTE, so in addition to providing access type permissions to 4 KiB pages, Smmpt34 supports 32 MiB megapages." However, while a L1 leaf MPTE does cover a 32 MiB address range (25 bits), I think the actual page size isn't 32 MiB. This is because the 3-bit NUMPGNRANGE range offset (as in  4.1.1.5) is used to select the specific page permission within that range. So, each permission tuple corresponds to a 4 MiB (25-3 bits) page size rather than the full 32 MiB range. The same hold for smmpt 43 52 64.

Linthereeder, 2025-04-24:
Also about smmptnapot (assuming I was correct about the previous issue): On page 22, it's stated that "When MPTE.L=1 and MPTE.N=1, the leaf MPTE represents a memory protection range that is part of a larger contiguous NAPOT memory protection range comprised of 512 pages for RV64 and 1024 pages for RV32." However, 512 pages of 4 KiB each in RV64 equals 2 MiB, which is identical to the L1 page size of 2 MiB in the Smmpt43 memory protection system. Given this, the smmptnapot extension doesn't seem to introduce any new granularity. Is it meant to be like this? Or have I misunderstand anything?

rsahita, 2025-05-07:
reg point 1 - your observation is correct - the spec should have said - the L1 leaf MPTE covers address space of X.

reg point 2 - as the note says - smmptnapot allows access-type permissions for one or more entries representing the contiguous region to be cached as a single cache entry as if it were a single (large) page.

Linthereeder, 2025-05-08:
Thanks for your reply. I understand that smmptnapot allows for caching a NAPOT range as a single entry, which can be helpful for performance.
However, my concern is not about caching efficiency, but about whether smmptnapot introduces any new granularity level between existing 4 KiB and 2 MiB levels.
Since 512 × 4 KiB = 2 MiB already corresponds to the L1 leaf MPTE granularity in Smmpt43, the napot range does not seem to expand the expressive granularity range.
Would it be more useful if the napot range could define a 128-page range (512 KiB), thereby introducing a new level of granularity that cannot be expressed in the current tree-based MPTE structure?

Linthereeder, 2025-05-08:
I mean, what is thel reason for a user to choose 4 KiB pages with the NAPOT extension, instead of simply using a single 2 MiB page?

Linthereeder, 2025-05-08:
Or perhaps we could use the reserved bits in the MPTE to indicate the NAPOT range size? That way, the extension could support flexible granularity between 4 KiB and 2 MiB, making it much more useful in practice?

Linthereeder, 2025-05-13:
I appreciate the explanation on the flexibility and efficiency of Smmptnapot. But I noticed that in the TLB domain, the simillar Svnapot extension introduces a new 64 KiB granularity, which is said to balance memory waste and fragmentation. This made me think that in the MPT domain, having a similar new granularity could also be beneficial. If the MPT could support addational granularity, it might offer similar advantages in terms of memory management as it is in TLB.

SiFiveHolland, 2025-05-16:
> Also about smmptnapot (assuming I was correct about the previous issue): On page 22, it's stated that "When MPTE.L=1 and MPTE.N=1, the leaf MPTE represents a memory protection range that is part of [...]

The difference is that with a L1 leaf MPTE, the entire 32 MiB range must use 2 MiB pages; but a group of L0 NAPOT MPTEs form a _single_ 2 MiB page, without restriction on the granularity of the adjacent pages. So the NAPOT MPTEs allow more flexibility, even though they do not add new page sizes.

HaoyuanLiuHZPH, 2025-05-19:
> The difference is that with a L1 leaf MPTE, the entire 32 MiB range must use 2 MiB pages; but a group of L0 NAPOT MPTEs form a _single_ 2 MiB page, without restriction on the granularity of the [...]

Thank you for helping clarify this point, I understand it now. I think it would be helpful if the specification could be updated with an explanation that while MPT NAPOT might initially appear to expand granularity akin to virtual pages, its actual role is in supporting a solitary 2M page. Thanks.

closed 2025-05-27 by rsahita