riscv/riscv-smmtt #141: questions about MFENCE Instruction opcode definition

Linthereeder, 2025-04-28:
I'm currently working on implementing the mpt, and It's seems that the opcode for mfence isn't defined anywhere. What are the specific values of the opcode, funct7 and funct3 fields for mfence? Have these values been officially settled yet?

Linthereeder, 2025-04-29:
And also, the mpt CSRs, mmpt and msdcfg, needs to define the assigning No.

rock-ifly, 2025-05-07:
We have the same question.

rsahita, 2025-05-07:
Hi @Linthereeder @rock-ifly the official CSR and opcode values will be assigned upon ARC approval before public review - until then please use any unassigned values tentatively.

rock-ifly, 2025-05-08:
> Hi [@Linthereeder](https://github.com/Linthereeder) [@rock-ifly](https://github.com/rock-ifly) the official CSR and opcode values will be assigned upon ARC approval before public review - until then please use any unassigned values tentatively.

@rsahita OK， thanks.

Linthereeder, 2025-05-08:
@rsahita great, thanks

closed 2025-05-12 by rsahita