riscv/riscv-smmtt #168: Question about IOMPT extension

1252005989, 2025-07-24:
Hi,
In the I/O MPT Extension section, it is mentioned: "The SRL and SML fields along with operand-1.SSM field are used to determine the effective RCID and MCID provided by the IOMMU for device originated requests. The determination of the effective RCID and MCID is as specified by Section 8.3. "

However, I still do not know what MPTCHK should perform. Could you explain it further?

Additionally, for the MCID/RCID provided by the IOMMU, what do these fields consist of? Do they include the prefixes of the RCID and MCID fields configured by RDSM, as well as the suffixes of the RCID and MCID configured by the Supervisor Domain?

ved-rivos, 2025-07-29:
> However, I still do not know what MPTCHK should perform. 

MPT check is checking memory accesses from the IOMMU and from devices associated with that IOMMu using the memort protection table. This is orthogonal to associating RCID/MCID with the memory accesses.

>Additionally, for the MCID/RCID provided by the IOMMU, what do these fields consist of? Do they include the prefixes of the RCID and MCID fields configured by RDSM, as well as the suffixes of the RCID and MCID configured by the Supervisor Domain? Are the MCID/RCID provided by the IOMMU not used directly for hardware operations?

They are used to generate the effective RCID and MCID - using algorithm in section 8.3 - for the memory accesses by the devices and IOMMU associated with a supervisor domain. In that sense they are similar to what the OS in a supervisor domain would configure into srmcfg on the hart.

1252005989, 2025-07-30:
> They are used to generate the effective RCID and MCID - using algorithm in section 8.3 - for the memory accesses by the devices and IOMMU associated with a supervisor domain. In that sense they are similar to what the OS in a supervisor domain would configure into srmcfg on the hart.

Yes, I think so. But I think IOMPT needs to provide the prefixes of the RCID and MCID fields configured by RDSM for each SD. Moreover, in the version 0.4 specification, I cannot find this field.

ved-rivos, 2025-07-30:
There is a update in progress to the 0.4 spec. The earlier intent was to have the QRID carried on the bus and the QoS controller to apply the prefixes but this is changing to instead not add more metadata on the bus.

ved-rivos, 2025-07-31:
The SRL, SML, and SQRID configurations are provided using the operand-0 register:
```
{reg: [
  {bits:  4, name: 'SRC_IDT (WARL)'},
  {bits:  2, name: 'SRC_IDM (WARL)'},
  {bits:  2, name: 'TEE_FLT (WARL)'}, 
  {bits: 24, name: 'SRC_ID'},
  {bits:  8, name: 'IOMMU_ID (WARL)'},
  {bits:  8, name: 'IOSDID (WARL)'},
  {bits:  4, name: 'SRL'},
  {bits:  4, name: 'SML'},
  {bits:  4, name: 'SQRID'},
  {bits:  4, name: 'WPRI'}
], config:{lanes: 8, hspace:1024}}

```
The `SSM` (now `SSRM` and `SSMM`) are specified using the operand-1 register:
```
{reg: [
  {bits:  4, name: 'MPT_MODE (WARL)'},
  {bits:  1, name: 'PPNV (WARL)'},
  {bits:  1, name: 'S (WARL)'},
  {bits:  1, name: 'IOSDIDV'},
  {bits:  1, name: 'SSM'},
  {bits:  2, name: 'WPRI'},
  {bits: 44, name: 'PPN'},
  {bits: 10, name: 'WPRI'}
], config:{lanes: 8, hspace:1024}}

```

The RDSM uses the `control` register to issue the `SET_ENTRY` operation to IOMPT. This operation uses operand-0 and operand-1 to create a classification rule `RULEID` to classify a device ID to the rule. The IOMMU itself if a DMA capable device i.e. when IOMMU walks page tables, walks DDT, walks PDT, reads command queue, writes fault queue, writes page request queue or generates MSI, etc. it makes memory accesses using the IOMMUs device ID. A rule needs to be created to associate the IOMMU device ID itself with an IOSDID, MPT, SQRID, SSM, SRL, SML, etc. 

For a DMA from device with ID=X the IOBRIDGE does following:
1. Invoke SDCL to classify the device ID=X to a rule. The SDCL returns the configurations for that rule. This includes the IOMMU_ID that will be used to translate that DMA. This also includes the SSM, SRL, SML, and SQRID.
2. The IObridge then invokes the IOMMU determined by IOMMU_ID to do the address translations.
3. The IOMMU may itself make memory accesses - such as to its page tables to perform the address translation. Let the IOMMU device ID be Y. The IOMMU accesses may include the RCID and MCID from the iommu_qosid register or from the device context.
4. The IOBRIDGE invokes MPTCHK using the MPT provided by SDCL to check the IOMMU memory access and if the access is allowed uses the SSM/SRL/SML/SQRID to determine effective RCID and MCID for the IOMMU memory access.
5. Eventually IOMMU provides translation for the DMA from device X along with the RCID and MCID from device context. The IOBRIDGE then invokes MPTCHK using the MPT provided by SDCL to check the translated address and if access is allowed uses the SSM/SRL/SML/SQRID to determine effective RCID and MCID for the device memory access.

zhuotianshu, 2025-07-31:
> 4. The IOBRIDGE invokes MPTCHK using the MPT provided by SDCL to check the IOMMU memory access and if the access is allowed uses the SSM/SRL/SML/SQRID to determine effective RCID and MCID for the IOMMU memory access.

Is the concrete algorithm for determining the effective RCID and MCID using SSM, SRL, SML, and SQRID still under development? It doesn't seem to have been included in yesterday's pull. (https://github.com/riscv/riscv-smmtt/pull/172)

1252005989, 2025-07-31:
> There is a update in progress to the 0.4 spec. The earlier intent was to have the QRID carried on the bus.

You mean there are no prefixes of the RCID and MCID fields from M-level (Machine level) configured in earlier intended usage? Will these fields be added in the future?

 Could you provide the latest release plan for the SMMTT specification?

ved-rivos, 2025-07-31:
> Is the concrete algorithm for determining the effective RCID and MCID using SSM, SRL, SML, and SQRID still under development? It doesn't seem to have been included in yesterday's pull request. (https://github.com/riscv/riscv-smmtt/pull/172)

Yes, sorry I have a commit missing there

1252005989, 2025-07-31:
> Could you provide the latest release plan for the SMMTT specification?

I find the release plan is not up-to-date in this URL
https://lf-riscv.atlassian.net/browse/RVS-1345

ved-rivos, 2025-07-31:
> Is the concrete algorithm for determining the effective RCID and MCID using SSM, SRL, SML, and SQRID still under development? It doesn't seem to have been included in yesterday's pull request.

Please take a look. I pushed and updated to PR #172

zhuotianshu, 2025-08-02:
> > Is the concrete algorithm for determining the effective RCID and MCID using SSM, SRL, SML, and SQRID still under development? It doesn't seem to have been included in yesterday's pull request.
> 
> Please take a look. I pushed and updated to PR [#172](https://github.com/riscv/riscv-smmtt/pull/172)

Thank you. I think this update can solve the problem above.

closed 2025-08-05 by ved-rivos